openUC2/UC2-REST #4: Troubleshooting cant compile with ps3/ps4Controller library

ranranking, 2022-02-23:
There are compile issue to use the ps3 or ps4Controller library. The Error is written as 'Error compiling for the board esp32 DEV Module'. The problem is the library only compatible with old version of the ESP libaray.
Solution is listed as in [here](https://forum.arduino.cc/t/error-compiling-for-board-esp32-dev-module-with-ps4controller-library/933405/2)

Short summary:
1. Go to the library folder, e.g. C:\Users\xxx\Documents\Arduino\libraries\PS4-esp32-master\src
2. Edit file **ps4_spp.c** comment out Line **86,87,88,90** such as  `//   esp_bt_gap_set_scan_mode(ESP_BT_CONNECTABLE, ESP_BT_NON_DISCOVERABLE);`
3. Edit file **PS4Controller.cpp**, add a line at line **44** with `#define ESP_BD_ADDR_STR         "%02hhx:%02hhx:%02hhx:%02hhx:%02hhx:%02hhx"`

Now you should be able to compile your code

beniroquai, 2022-02-24:
Hey @ranranking I trie to understand what happens :D .
So the current code in this repo does not yet work with the PS4 controller it seems. Time to change this. Ok. You used this [library](https://github.com/aed3/PS4-esp32) I assume? 

I have updated the library [here](https://github.com/beniroquai/PS4-esp32/). Can you check if this works?

ranranking, 2022-02-24:
Hey @beniroquai I checked the library, it seems you already modified the code i mentioned there. The library is the one i used, only one small mistake, in **ps4_spp.c** the line **89** should not be comment out. Thats why i wrote them separately. :) Other than that is everything fine

beniroquai, 2022-02-24:
Great. Would you mind having a look at this program and try it out with your current system?
[https://github.com/openUC2/UC2-REST/tree/master/ESP32/REST_API_JSON_Serial_Wifi_motor_PS3_v0](https://github.com/openUC2/UC2-REST/tree/master/ESP32/REST_API_JSON_Serial_Wifi_motor_PS3_v0). It should work with PS4 controllers. Also if something is unclear, please add any comments so that I can add it to the documentation! :-)

ranranking, 2022-02-25:
It works with my controller now. There are three points i found:
1. For compiling the library from Adafruit NeoMatrix and its dependency are required, clarify it on the library section?
2. Laser_pin is not assigned on any key from the controller, i used analogout_pin for laser. Matching the pindef?
3. It might missing enable the motor when using controller, currently i used `digitalWrite(ENABLE,LOW);` at Line 234 in `run_motor()` from **control_PS4.ino**. Add enable at somewhere with your style?

beniroquai, 2022-02-25:
Hey @ranranking Thanks for trying it out! 

1. Can you specify the names for the library again? You could also change the README.md file directly, I guess? ;) 
2. True, the layout for the controller is not really fixed. You can adapt it. Do you have any good way to "map" the functions to the controller? Analog out works too, but perhaps we can have that more generic? 
3. That seems to be a bug indeed, I've fixed it and wondered how it worked in the first place :D ..