openUC2/UC2-REST #7: ImSwitch troubleshooting

ranranking, 2022-03-08:
I got that error during boost imswitch, do you have any idea what is the problem?
```
During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\wanghaoran\.conda\envs\imswitch\lib\site-packages\imswitch\imcontrol\controller\ImConMainController.py", line 135, in closeEvent
    self.__factory.closeAllCreatedControllers()
AttributeError: 'ImConMainController' object has no attribute '_ImConMainController__factory'

2022-03-08 13:05:15 ERROR [main] Failed to initialize module imcontrol
2022-03-08 13:05:15 ERROR [main] Traceback (most recent call last):
  File "c:\Users\wanghaoran\Desktop\ImSwitch\ImSwitch-master\imswitch\__main__.py", line 50, in main
    view, controller = modulePkg.getMainViewAndController(
  File "C:\Users\wanghaoran\.conda\envs\imswitch\lib\site-packages\imswitch\imcontrol\__init__.py", line 55, in getMainViewAndController
    raise e
  File "C:\Users\wanghaoran\.conda\envs\imswitch\lib\site-packages\imswitch\imcontrol\__init__.py", line 52, in getMainViewAndController
    controller = ImConMainController(options, setupInfo, view, moduleCommChannel)
  File "C:\Users\wanghaoran\.conda\envs\imswitch\lib\site-packages\imswitch\imcontrol\controller\ImConMainController.py", line 35, in __init__
    self.__masterController = MasterController(self.__setupInfo, self.__commChannel,
  File "C:\Users\wanghaoran\.conda\envs\imswitch\lib\site-packages\imswitch\imcontrol\controller\MasterController.py", line 21, in __init__
    self.rs232sManager = RS232sManager(self.__setupInfo.rs232devices)
  File "C:\Users\wanghaoran\.conda\envs\imswitch\lib\site-packages\imswitch\imcontrol\model\managers\RS232sManager.py", line 12, in __init__
    super().__init__(rs232deviceInfos, 'rs232', **lowLevelManagers)
  File "C:\Users\wanghaoran\.conda\envs\imswitch\lib\site-packages\imswitch\imcontrol\model\managers\MultiManager.py", line 17, in __init__
    package = importlib.import_module(
  File "C:\Users\wanghaoran\.conda\envs\imswitch\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 973, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'imswitch.imcontrol.model.managers.rs232.ESP32Manager'
```

beniroquai, 2022-03-09:
May I ask you which version of ImSwitch you use? 
Can you also post the current config-`json` file that describes the system? 

Did you use our [fork](https://github.com/beniroquai/ImSwitch)?

ranranking, 2022-03-09:
I have `ImSwitch   1.2.1`
`json` file you mean `imcontrol_options`? I overwrote the whole folder with the github repo
```
{
    "setupFileName": "example_uc2_multicolour.json",
    "recording": {
        "outputFolder": "./ImSwitch/ImSwitch/recordings",
        "includeDateInOutputFolder": true
    },
    "setupFileName10": "example_uc2_lightsheet_rotating.json",
    "setupFileName0": "example_uc2_STORM_Berlin.json",
    "setupFileName11": "example_uc2_xyz_only.json",
    "setupFileName1": "example_uc2_lightsheet_vimba.json",
    "setupFileName7": "example_uc2_multicolour.json",
    "setupFileName4": "example_uc2_lightsheet_vimba.json",
    "setupFileName3": "example_uc2_multicolour_basler.json",
    "setupFileName2": "example_uc2_.json"
}
```

ranranking, 2022-03-10:
I tried to install the UC2 version of imswitch, now it works. Could you only show how to install UC2 version of ImSwitch [there](https://github.com/beniroquai/ImSwitch) and maybe mention and cite the original imswitch repo? It dismissed me to install it with the earlier options using `pip install imswitch`, i thought three options all work

beniroquai, 2022-03-16:
I think we should do another Hackathon for that with Xavier ;-)

closed 2025-04-22 by beniroquai