openUC2/UC2-REST #9: Error when starting imswitch

AZehrer, 2022-08-23:
Traceback (most recent call last):
  File "C:\Users\Ando\Anaconda3\envs\imswitch\Scripts\imswitch-script.py", line 33, in <module>
    sys.exit(load_entry_point('ImSwitch', 'console_scripts', 'imswitch')())
  File "C:\Users\Ando\Anaconda3\envs\imswitch\Scripts\imswitch-script.py", line 25, in importlib_load_entry_point
    return next(matches).load()
  File "C:\Users\Ando\Anaconda3\envs\imswitch\lib\importlib\metadata.py", line 77, in load
    module = import_module(match.group('module'))
  File "C:\Users\Ando\Anaconda3\envs\imswitch\lib\importlib\_init_.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 843, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "c:\users\ando\documents\imswitch_dev\imswitch\_main_.py", line 6, in <module>
    from imswitch.imcommon.controller import ModuleCommunicationChannel, MultiModuleWindowController
  File "c:\users\ando\documents\imswitch_dev\imswitch\imcommon\controller\_init_.py", line 2, in <module>
    from .MultiModuleWindowController import MultiModuleWindowController
  File "c:\users\ando\documents\imswitch_dev\imswitch\imcommon\controller\MultiModuleWindowController.py", line 9, in <module>
    from .CheckUpdatesController import CheckUpdatesController
  File "c:\users\ando\documents\imswitch_dev\imswitch\imcommon\controller\CheckUpdatesController.py", line 5, in <module>
    from luddite import get_version_pypi
  File "C:\Users\Ando\Anaconda3\envs\imswitch\lib\site-packages\luddite.py", line 11, in <module>
    from pkg_resources import parse_version
  File "C:\Users\Ando\Anaconda3\envs\imswitch\lib\site-packages\pkg_resources\_init_.py", line 3243, in <module>
    def _initialize_master_working_set():
  File "C:\Users\Ando\Anaconda3\envs\imswitch\lib\site-packages\pkg_resources\_init_.py", line 3226, in _call_aside
    f(*args, **kwargs)
  File "C:\Users\Ando\Anaconda3\envs\imswitch\lib\site-packages\pkg_resources\_init_.py", line 3255, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "C:\Users\Ando\Anaconda3\envs\imswitch\lib\site-packages\pkg_resources\_init_.py", line 568, in _build_master
    ws.require(_requires_)
  File "C:\Users\Ando\Anaconda3\envs\imswitch\lib\site-packages\pkg_resources\_init_.py", line 886, in require
    needed = self.resolve(parse_requirements(requirements))
  File "C:\Users\Ando\Anaconda3\envs\imswitch\lib\site-packages\pkg_resources\_init_.py", line 772, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'openflexure-microscope-client' distribution was not found and is required by ImSwitch

(imswitch39) C:\Users\Ando>pip install openflexuremicroscope-client
ERROR: Could not find a version that satisfies the requirement openflexuremicroscope-client (from versions: none)
ERROR: No matching distribution found for openflexuremicroscope-client

beniroquai, 2022-08-23:
Thanks @AZehrer for reporting the issue. Indeed the `setup.cfg` lacks the dependency. 

Please try installing it by entering:
`pip install openflexure-microscope-client`

I'll fix the dependencies.

beniroquai, 2022-08-23:
Fixed in the latest commit

closed 2022-08-23 by beniroquai

AZehrer, 2022-08-23:
Still got this superlong error:

Building wheel for Pillow (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: 'C:\Users\Ando\Anaconda3\envs\imswitch39\python.exe' -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\Ando\\AppData\\Local\\Temp\\pip-install-lrluxrxk\\pillow_99c264feb9f642738c22338ec1b890de\\setup.py'"'"'; __file__='"'"'C:\\Users\\Ando\\AppData\\Local\\Temp\\pip-install-lrluxrxk\\pillow_99c264feb9f642738c22338ec1b890de\\setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d 'C:\Users\Ando\AppData\Local\Temp\pip-wheel-pi4mfelf'
       cwd: C:\Users\Ando\AppData\Local\Temp\pip-install-lrluxrxk\pillow_99c264feb9f642738c22338ec1b890de\
  Complete output (181 lines):
  C:\Users\Ando\AppData\Local\Temp\pip-install-lrluxrxk\pillow_99c264feb9f642738c22338ec1b890de\setup.py:42: RuntimeWarning: Pillow 7.2.0 does not support Python 3.9 and does not provide prebuilt Windows binaries. We do not recommend building from source on Windows.
    warnings.warn(
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build\lib.win-amd64-3.9
  creating build\lib.win-amd64-3.9\PIL
  copying src\PIL\BdfFontFile.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\BlpImagePlugin.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\BmpImagePlugin.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\BufrStubImagePlugin.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\ContainerIO.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\CurImagePlugin.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\DcxImagePlugin.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\DdsImagePlugin.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\EpsImagePlugin.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\ExifTags.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\features.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\FitsStubImagePlugin.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\FliImagePlugin.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\FontFile.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\FpxImagePlugin.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\FtexImagePlugin.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\GbrImagePlugin.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\GdImageFile.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\GifImagePlugin.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\GimpGradientFile.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\GimpPaletteFile.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\GribStubImagePlugin.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\Hdf5StubImagePlugin.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\IcnsImagePlugin.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\IcoImagePlugin.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\Image.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\ImageChops.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\ImageCms.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\ImageColor.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\ImageDraw.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\ImageDraw2.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\ImageEnhance.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\ImageFile.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\ImageFilter.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\ImageFont.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\ImageGrab.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\ImageMath.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\ImageMode.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\ImageMorph.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\ImageOps.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\ImagePalette.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\ImagePath.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\ImageQt.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\ImageSequence.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\ImageShow.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\ImageStat.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\ImageTk.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\ImageTransform.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\ImageWin.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\ImImagePlugin.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\ImtImagePlugin.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\IptcImagePlugin.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\Jpeg2KImagePlugin.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\JpegImagePlugin.py -> build\lib.win-amd64-3.9\PIL
  copying src\PIL\JpegPresets.py -> build\lib.win-amd64-3.9\PIL
[... 301 lines cut ...]
    __main__.RequiredDependencyException: zlib

    During handling of the above exception, another exception occurred:

    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "C:\Users\Ando\AppData\Local\Temp\pip-install-lrluxrxk\pillow_99c264feb9f642738c22338ec1b890de\setup.py", line 918, in <module>
        raise RequiredDependencyException(msg)
    __main__.RequiredDependencyException:

[... cut ...] pen)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\Ando\AppData\Local\Temp\pip-record-t5ql8u8g\install-record.txt' --single-version-externally-managed --compile --install-headers 'C:\Users\Ando\Anaconda3\envs\imswitch39\Include\Pillow' Check the logs for full command output.

(imswitch39) C:\Users\Ando>

beniroquai, 2022-08-23:
What did you enter in the first place? Just starting imswitch?

AZehrer, 2022-08-24:
At the end of the error, this was written:
The headers or library files could not be found for zlib,
a required dependency when compiling Pillow from source.

Please see the install instructions at:
https://pillow.readthedocs.io/en/latest/installation.html

which i did, and then tried following comand again, which gave the error:
pip install openflexure-microscope-client

beniroquai, 2022-08-25:
I had this error a while back. I absolutely don't no what'S the reason for this. I think i was able to get rid of it by typing: 

```

pip install ./ -e --no-deps Pillow
```

and then 

```
pip install Pillow
```