openUC2/UC2-REST #14: Better logic for Wifi (e.g. setup in background)

beniroquai, 2022-08-28:
Some thoughts:

What would be a good way to integrate the ESP32 into an 
1. existing network ?
2. connect to an AP?
3. use a captive portal?

My take on that:

If not already connected to a predefined network, we should offer a captive portal / wifimanager that helps the user to connect to an existing network. If this fails / or the user decides to use the ESP32 as an Access Point, we will go ahead. 
The webserver will always run and offer stuff.

ranranking, 2022-08-29:
easiest way is maybe include a prenamed wifi stick into the box? It costs like some euro and they can easily add every parts on it?

beniroquai, 2022-08-29:
The esp32 can act as a wifi access point without problem. But it lacks
internet access.

beniroquai, 2022-09-03:
Perhaps the ESP32 starts with a known access point (e.g. UC2/youseetoo) and keeps the captive portal open in order to change the credentials for the local (internetnable) network? 
@renerichter ?

closed 2024-02-20 by beniroquai