openUC2/UC2-REST #19: Imscripting issue with the LED control

AZehrer, 2022-09-29:
The LED can be controlled in imswitch, however when using an imscript code, the LED won't turn on with following command:
api.imcontrol.setLEDAllOn () 

Even though this command used to work a few weeks ago.

kasasxav, 2022-09-29:
Hey! So that's to generate the documentation, I think it should work. It's just not uploaded to readthedocs until there is a release.

What is exactly not working? Do you have an error message?

beniroquai, 2022-09-29:
I'm getting the following error (which is obviously a UC2-related error). I feel I forgot to initialize something somewhere..

```
2022-09-29 22:51:56 DEBUG [ImConMainController] Initializing
2022-09-29 22:51:59 DEBUG [ImConMainController] Started server with URI -> PYRO:ImSwitchServer@:::54333
Object <imswitch.imcontrol.controller.server.ImSwitchServer.ImSwitchServer object at 0x15438c9d0>:
    uri = PYRO:ImSwitchServer@[::]:54333
Pyro daemon running.
Backend QtAgg is interactive backend. Turning interactive mode on.
2022-09-29 22:52:08 DEBUG [ImConMainController] Shutting down

E+01034.163: Failed to kill Debuggee[PID=38653]
             
             Traceback (most recent call last):
               File "/Users/bene/.vscode/extensions/ms-python.python-2022.14.0/pythonFiles/lib/python/debugpy/adapter/../../debugpy/launcher/../../debugpy/launcher/debuggee.py", line 190, in kill
                 os.killpg(process.pid, signal.SIGKILL)
             PermissionError: [Errno 1] Operation not permitted
             
             Stack where logged:
               File "/Users/bene/.vscode/extensions/ms-python.python-2022.14.0/pythonFiles/lib/python/debugpy/adapter/../../debugpy/launcher/../../debugpy/launcher/debuggee.py", line 192, in kill
                 log.swallow_exception("Failed to kill {0}", describe())
               File "/Users/bene/.vscode/extensions/ms-python.python-2022.14.0/pythonFiles/lib/python/debugpy/adapter/../../debugpy/launcher/../../debugpy/common/log.py", line 215, in swallow_exception
                 _exception(format_string, *args, **kwargs)
             

(imswitch39) Benedicts-MacBook-Pro:ImSwitch bene$ 
(imswitch39) Benedicts-MacBook-Pro:ImSwitch bene$  cd /Users/bene/Dropbox/Dokumente/Promotion/PROJECTS/MicronController/PYTHON/ImSwitch ; /usr/bin/env /opt/anaconda3/envs/imswitch39/bin/python /Users/bene/.vscode/extensions/ms-python.python-2022.14.0/pythonFiles/lib/python/debugpy/adapter/../../debugpy/launcher 61582 -- /Users/bene/Dropbox/Dokumente/Promotion/PROJECTS/MicronController/PYTHON/ImSwitch/tools/generateapidocs.py 
No module named 'javabridge'
Problem setting classpath. Switching to conventional viewer by setting __DEFAULTS__['IMG_VIEWER'] = 'NIP_VIEW' 
2022-09-29 23:08:10 DEBUG [imcontrol init] Setup used: 
2022-09-29 23:08:10 DEBUG [ImConMainView] Initializing
/Users/bene/Dropbox/Dokumente/Promotion/PROJECTS/MicronController/PYTHON/ImSwitch/imswitch/imcommon/view/guitools/naparitools.py:97: FutureWarning: Public access to Window.qt_viewer is deprecated and will be removed in
v0.5.0. It is considered an "implementation detail" of the napari
application, not part of the napari viewer model. If your use case
requires access to qt_viewer, please open an issue to discuss.
  napariViewer.window.qt_viewer.dockLayerList
/Users/bene/Dropbox/Dokumente/Promotion/PROJECTS/MicronController/PYTHON/ImSwitch/imswitch/imcommon/view/guitools/naparitools.py:100: FutureWarning: Public access to Window.qt_viewer is deprecated and will be removed in
v0.5.0. It is considered an "implementation detail" of the napari
application, not part of the napari viewer model. If your use case
requires access to qt_viewer, please open an issue to discuss.
  napariViewer.window.qt_viewer.dockLayerList.qt_area,
/Users/bene/Dropbox/Dokumente/Promotion/PROJECTS/MicronController/PYTHON/ImSwitch/imswitch/imcommon/view/guitools/naparitools.py:101: FutureWarning: Public access to Window.qt_viewer is deprecated and will be removed in
v0.5.0. It is considered an "implementation detail" of the napari
application, not part of the napari viewer model. If your use case
requires access to qt_viewer, please open an issue to discuss.
  napariViewer.window.qt_viewer.dockLayerList
/Users/bene/Dropbox/Dokumente/Promotion/PROJECTS/MicronController/PYTHON/ImSwitch/imswitch/imcommon/view/guitools/naparitools.py:103: FutureWarning: Public access to Window.qt_viewer is deprecated and will be removed in
v0.5.0. It is considered an "implementation detail" of the napari
application, not part of the napari viewer model. If your use case
requires access to qt_viewer, please open an issue to discuss.
  napariViewer.window.qt_viewer.dockLayerList.show()
/Users/bene/Dropbox/Dokumente/Promotion/PROJECTS/MicronController/PYTHON/ImSwitch/imswitch/imcontrol/view/widgets/ImageWidget.py:94: FutureWarning: Public access to Window.qt_viewer is deprecated and will be removed in
v0.5.0. It is considered an "implementation detail" of the napari
application, not part of the napari viewer model. If your use case
requires access to qt_viewer, please open an issue to discuss.
  canvas=self.napariViewer.window.qt_viewer.canvas,
/Users/bene/Dropbox/Dokumente/Promotion/PROJECTS/MicronController/PYTHON/ImSwitch/imswitch/imcontrol/view/widgets/ImageWidget.py:95: FutureWarning: Public access to Window.qt_viewer is deprecated and will be removed in
v0.5.0. It is considered an "implementation detail" of the napari
application, not part of the napari viewer model. If your use case
requires access to qt_viewer, please open an issue to discuss.
  view=self.napariViewer.window.qt_viewer.view,
/Users/bene/Dropbox/Dokumente/Promotion/PROJECTS/MicronController/PYTHON/ImSwitch/imswitch/imcontrol/view/widgets/ImageWidget.py:96: FutureWarning: Public access to Window.qt_viewer is deprecated and will be removed in
v0.5.0. It is considered an "implementation detail" of the napari
application, not part of the napari viewer model. If your use case
[... cut ...]
    return self.ESP32.config.loadConfigDevice()
AttributeError: 'UC2ConfigManager' object has no attribute 'ESP32'
```

Happening here: https://github.com/openUC2/ImSwitch/blob/master/imswitch/imcontrol/model/managers/UC2ConfigManager.py#L46

It seems the ESP32 manager is not initialized:
https://github.com/openUC2/ImSwitch/blob/master/imswitch/imcontrol/model/managers/UC2ConfigManager.py#L29

Which is really weird, since this would mean that the `__init__` method is never called. Any clue?! :D

beniroquai, 2022-09-29:
But next question: How would I access the ESP32Manager in the correct way? https://github.com/openUC2/ImSwitch/blob/master/imswitch/imcontrol/model/managers/UC2ConfigManager.py#L29

closed 2024-02-20 by beniroquai