openUC2/UC2-REST #76: Difference in code for Arduino IDE

jsmithe2, 2024-04-25:
I was following your set-up for how to add the R32 board to the Arduino IDE on youtube. I downloaded your code from the repository and opened the main.ipo file but it does not have the same code currently than what is displayed in the video. Specifically the pindef_incubator_microscope_zonly_matrix.h code is not there so am not able to follow the video beyond this point. What are your suggestions for me of possible troubleshooting?

beniroquai, 2024-05-01:
Hey @jsmithe2, sorry for my late reply. I have moved to a new firmware which can be flashed through this website: 
https://youseetoo.github.io/
and used through this website:
https://youseetoo.github.io/indexWebSerialTest.html
it only runs on ESP32's, I don't support Arduinos anymore. 

I hope this helps.

jsmithe2, 2024-05-02:
@beniroquai So theoretically, if I add the xy stepper motors from the neje laser printer to the esp32 board along with the z axis stepper motor and flash the firmware you provided here and confirm it to work on the web serial tester, it should be able to be used now with the esp32 stage manager code you wrote for the imswitch program. Is that correct thinking? Is there any potential problems you see that I might have?

closed 2024-06-18 by beniroquai