openUC2/UC2-REST #77: ESP 32 keeps rebooting and calls abort function

Farghal1, 2024-05-17:
Hello,
In the serial monitor I keep getting this same output. I think the issue is  with the reset/abort pin, but I'm not really sure about it or how to fix it.

Is it a hardware issue where this pin needs to be connected or is it something with the configuration in the code itself?

ELF file SHA256: 3cd5fedda79b99fe

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13260
load:0x40080400,len:3028
entry 0x400805e4
Start
You can use this software by sending JSON strings, A full documentation can be found here:
https://github.com/openUC2/UC2-REST/
/state_act
/state_get
/state_set

abort() was called at PC 0x40085293 on core 1

Backtrace: 0x40083d4d:0x3ffc9950 0x40093d55:0x3ffc9970 0x40099009:0x3ffc9990 0x40085293:0x3ffc9a10 0x4008

beniroquai, 2024-05-17:
Hey! Good catch. Which software did you use? I should remove any ESP-related stuff from here. You should have a look here: https://youseetoo.github.io/

Farghal1, 2024-05-22:
Hello,
Sorry for late reply. I still didn't use any software, I'm trying to flash the esp32. After flashing it, this is the output I get on the Serial monitor. I don't know what causes this issue. Is it software or hardware issue? And how to solve it?

Also, I tried the webpage provided on the link you sent above and I get the same output.

Can you please tell me how to proceed?

Farghal1, 2024-05-22:
Issue solved using by flashing the ESP32 using the updated version called UC2-ESP is implemented in the Platform.io framework.
 
One thing to know is the other technique didn't work because the program required 125% memory of ESP32 WEMOS D1 R32.

Farghal1, 2024-05-22:
When I try to send any command to the hardware, this is the output I get.

---- Sent utf8 encoded message: "{\"task\":\"/motor_act\", \"motor\": { \"steppers\": [ { \"stepperid\": 1, \"position\": 4800, \"speed\": 20000, \"isabs\": 0, \"isaccel\":0, \"accel\":100000} ] } }" ----
[ 84315][I][SerialProcess.cpp:31] loop(): String s:{"task":"/motor_act", "motor": { "steppers": [ { "stepperid": 1, "position": 4800, "speed": 20000, "isabs": 0, "isaccel":0, "accel":100000} ] } } , char:{"task":"/motor_act", "motor": { "steppers": [ { "stepperid": 1, "position": 4800, "speed": 20000, "isabs": 0, "isaccel":0, "accel":100000} ] } }
[ 84334][I][SerialProcess.cpp:158] jsonProcessor(): process motor act
[ 84340][I][FocusMotor.cpp:189] act(): motor act
[ 84344][E][Module.cpp:16] getJsonInt(): json value qid is null
[ 84350][E][Module.cpp:16] getJsonInt(): json value isen is null
[ 84356][E][Module.cpp:16] getJsonInt(): json value isforever is null
[ 84362][I][FocusMotor.cpp:325] act(): start stepper (act): motor:1 isforver:0, speed: 20000, maxSpeed: 40000, target pos: 4800, isabsolute: 0, isacceleration: 0, acceleration: 100000
[ 84366][E][FocusMotor.cpp:439] setExternalPin(): i2c write failed
++
{
        "qid":  0,
        "success":      1
}[ 84388][E][FocusMotor.cpp:439] setExternalPin(): i2c write failed

I don't know if this is a successful output or not, but no actions happen on the Hardware (Motors, LEDs, Laser, etc..).

beniroquai, 2024-05-22:
Did you flash via the website or PIO? The v3 board has a port extender that's connected via I2C, so you should rather use v2 (e.g. `UC2_2` which you set in the `PinConfig.h`)

Farghal1, 2024-05-22:
I flashed using PIO. How can I set it to be in the PinConfig.h?

beniroquai, 2024-05-22:
Change the value here: https://github.com/youseetoo/uc2-esp32/blob/main/main/PinConfig.h#L870

Farghal1, 2024-05-22:
I changed it to be UC2_2 and UC2_WEMOS and UC2_1.
[image]

I get errors while building the platformio.ini nothing seems to be built fine except UC2_3.

beniroquai, 2024-05-22:
Oh, interesting. Yeah, you have to add these to the default struct. You can change this and file a PR :)

beniroquai, 2024-05-22:
> Yeah Great. I added them to the default struct and the build finished
> successfully.

Feel free to perform a PR

Farghal1, 2024-06-03:
Hello!

The firmware is flashed successfully and when I try to send some commands to the hardware I get this success output, but I no effect on the hardware. The leds doesn't turn on and the motors doesn't move. what could be the issue?

Sent utf8 encoded message: "{\"task\": \"/ledarr_act\",\"LEDArrMode\": \"full\", \"red\":100, \"green\": 100, \"blue\":100}   {\"task\": \"/ledarr_act\",\"LEDArrMode\": \"full\", \"red\":0, \"green\": 0, \"blue\":0}   {'red': 193, 'green': 193, 'blue': 193, 'indexled': 27, 'Nleds': 1, 'LEDArrMode': 'single', 'task': '/ledarr_act'}   {'red': 193, 'green': 193, 'blue': 193, 'LEDArrMode': 'left', 'task': '/ledarr_act'}   {'red': 193, 'green': 193, 'blue': 193, 'LEDArrMode': 'right', 'task': '/ledarr_act'}   {'red': 193, 'green': 193, 'blue': 193, 'LEDArrMode': 'top', 'task': '/ledarr_act'}   {'red': 193, 'green': 193, 'blue': 193, 'LEDArrMode': 'bottom', 'task': '/ledarr_act'}" ----
[507867][I][SerialProcess.cpp:31] loop(): String s:{"task": "/ledarr_act","LEDArrMode": "full", "red":100, "green": 100, "blue":100}   {"task": "/ledarr_act","LEDArrMode": "full", "red":0, "green": 0, "blue":0}   {'red': 193, 'green': 193, 'blue': 193, 'indexled': 27, 'Nleds': 1, 'LEDArrMode': 'single', 'task': '/ledarr_act'}   {'red': 193, 'green': 193, 'blue': 193, 'LEDArrMode': 'left', 'task': '/ledarr_act'}   {'red': 193, 'green': 193, 'blue': 193, 'LEDArrMode': 'right', 'task': '/ledarr_act'}   {'red': 193, 'green': 193, 'blue': 193, 'LEDArrMode': 'top', 'task': '/ledarr_act'}   {'red': 193, 'green': 193, 'blue': 193, 'LEDArrMode': 'bottom', 'task': '/ledarr_act'} , char:{"task": "/ledarr_act","LEDArrMode": "full", "red":100, "green": 100, "blue":100}   {"task": "/ledarr_act","LEDArrMode": "full", "red":0, "green": 0, "blue":0}   {'red': 193, 'green': 193, 'blue': 193, 'indexled': 27, 'Nleds': 1, 'LEDArrMode': 'single', 'task': '/ledarr_act'}   {'red': 193, 'green': 193, 'blue': 193, 'LEDArrMode': 'left', 'task': '/ledarr_act'}   {'red': 193, 'green': 193, 'blue': 193, 'LEDArrMode': 'right', 'task': '/ledarr_act'}   {'red': 193, 'green': 193, 'blue': 193, 'LEDArrMode': 'top', 'task': '/ledarr_act'}   {'red': 193, 'green': 193, 'blue': 193, 'LEDArrMode': 'bottom', 'task': '/ledarr_act'}
[507970][E][Module.cpp:16] getJsonInt(): json value qid is null
++
{
        "qid":  0,
        "success":      1
}

--

beniroquai, 2024-06-03:
How did you wire up your hardware? Any schematics?

Farghal1, 2024-06-03:
[image]
This is the hardware schematic I used.

        "identifier_name":      "UC2_Feather",
        "identifier_id":        "V2.0",
        "identifier_date":      "Jun  3 202413:18:13",
        "identifier_author":    "BD",
        "IDENTIFIER_NAME":      "uc2-esp",
        "configIsSet":  0,
        "pindef":       "UC2_2",
        "qid":  0
}
--

This is also the output of excuting state_get

beniroquai, 2024-06-03:
Can you try flashing the WEMOS example from the you see too website? Which esp board are you using?

Farghal1, 2024-06-03:
I mean if the above output that I get when I excute state_get is correct, so the flashing is correct and the esp behaves as desired and it's only hardware issue, right?

beniroquai, 2024-06-03:
Ok @AZehrer you could help here, too? ;) 

Please choose this board and see what's happening:
[image]

Farghal1, 2024-06-03:
I can communicate effectively with the ESP through the serial monitor, but nothing performs of the hardware.

beniroquai, 2024-06-03:
Can you try to flash the hardware without the CNC board attached?
when you do the `status_get` command the board's firmware should be
UC2_WEMOS not UC2_2

Farghal1, 2024-06-03:
No I get the state as "pindef": UC2_2

Farghal1, 2024-06-03:
Issue solved by changing the pindef to UC2_WEMOS as there the pin configuration for each one is different.

closed 2024-06-05 by beniroquai