openUC2/UC2-REST #79: Laser not working

Farghal1, 2024-07-08:
Hello,

I'm trying to control the laser using imswitch. The outpin SpnEn doesn't seem to have any output voltage and the laser isn't responding to any change in intensity from imswitch.

I'm using wemos d1 r32 with these pin configurations.

struct UC2_WEMOS : PinConfig // also used for cellSTORM wellplateformat
{
     const char *pindefName = "UC2_WEMOS";
     const char *mSSIDAP = "UC2_WEMOS";
     const char *mSSID = "UC2_WEMOS";
     const char *mPWD = "12345678";

     int pwm_resolution = 12; //8bit 256, 10bit  1024, 12bit 4096;
     int pwm_frequency =  19000; // 5000; //12000
     
     bool mAP = true;
     const char *hostname = "youseetoo";

     // ESP32-WEMOS D1 R32
     int8_t MOTOR_A_DIR = GPIO_NUM_23; // Bridge from Endstop Z to Motor A (GPIO_NUM_23)
     int8_t MOTOR_X_DIR = GPIO_NUM_16;
     int8_t MOTOR_Y_DIR = GPIO_NUM_27;
     int8_t MOTOR_Z_DIR = GPIO_NUM_14;
     int8_t MOTOR_A_STEP = GPIO_NUM_5; // Bridge from Endstop Y to Motor A (GPIO_NUM_5)
     int8_t MOTOR_X_STEP = GPIO_NUM_26;
     int8_t MOTOR_Y_STEP = GPIO_NUM_25;
     int8_t MOTOR_Z_STEP = GPIO_NUM_17;
     int8_t MOTOR_ENABLE = GPIO_NUM_12;
     bool MOTOR_ENABLE_INVERTED = true;

     int8_t LASER_1 = GPIO_NUM_18; // WEMOS_D1_R32_SPINDLE_ENABLE_PIN
     int8_t LASER_2 = GPIO_NUM_19; // WEMOS_D1_R32_SPINDLEPWMPIN
     int8_t LASER_3 = GPIO_NUM_13; // WEMOS_D1_R32_X_LIMIT_PIN

     int8_t LED_PIN = GPIO_NUM_4;
     int8_t LED_COUNT = 64;

     int8_t PIN_DEF_END_X = 0;
     int8_t PIN_DEF_END_Y = disabled; // GPIO_NUM_5;
     int8_t PIN_DEF_END_Z = disabled; // GPIO_NUM_23;

     const char *PSX_MAC = "00:32:94:b9:10:73";
     int8_t PSX_CONTROLLER_TYPE = 2;
     boolean enableBlueTooth = true;

     int8_t JOYSTICK_SPEED_MULTIPLIER = 5;
     int8_t JOYSTICK_SPEED_MULTIPLIER_Z = 3;

};

the following are imswitch configurations.

"lasers":{
"635 Laser": {
  "analogChannel": null,
  "digitalLine": null,
  "managerName": "ESP32LEDLaserManager",
  "managerProperties": {
      "rs232device": "ESP32",
      "channel_index": 1, 
      "filter_change": true
  },
  "wavelength": 635,
  "valueRangeMin": 0,
  "valueRangeMax": 255
},

Am I missing something or is there any extra manipulation that I need to do for the laser to make it work?

beniroquai, 2024-07-08:
Is it flashing briefly when it's booting up?

> Message ID: ***@***.***>
>

Farghal1, 2024-07-11:
No, It's always ON and when the spnEN is connected it's OFF.

beniroquai, 2024-07-11:
how did you wire GND/Signal? 12V/GND? There is a rectifier in most cases,
so you would measure against virtual GND

Farghal1, 2024-07-11:
[image]

This is the schematic I used. Is this the correct one?

beniroquai, 2024-07-11:
As I said, the Laser board has a rectifier after the 12V, this has to be connected in a proper way (usually not indicated). 
Can you turn on/off the laser with 5V/GND signals?

Farghal1, 2024-07-12:
No the 5V signal has no effect on the laser.

beniroquai, 2024-07-12:
Reverse 12v Input and check again. Maybe laser is broken

Farghal1, 2024-07-15:
The laser is working fine, but the spnEN has no output voltage. Could this be an issue with the firmware?

beniroquai, 2024-07-15:
Can you check that pin without the CNC Shield connected? May also be that the ESP’s pin is fried.

jsmithe2, 2024-07-16:
@Farghal1 what power supply are you using for your setup? I have the exact same setup except for the laser module and my motors are not working properly. I am thinking I might not be running the proper power to the board for all the motors to work.

beniroquai, 2024-07-16:
Did you bridge the 12 v from esp to CNC?

jsmithe2, 2024-07-16:
@beniroquai So I was trying to avoiding soldering so I connected two different power supplies vis barrel joack to both the esp board and the CNC shield. I not realizing I might not be supplying enough power.

I am using a NEJE laser engraver and the already assembled nema 17 motors for the X and Y. I Just used to power cord it came with to power the CNC shield. I used a red and black wire to barrel jack adapter to connected it to the shield FYI. Not sure if that is okay. Both barrel jack power supplies connected to the cnc shield and the esp board have a 12V/ 3A output rating. 

Is this not enough power for the 2 nema 17 and 1 nema 11 stepper motors? Do you have a recommended power supply, and do you think soldering a bridge from the esp and CNC is required?

beniroquai, 2024-07-17:
12V @ 3A should be fairly enough actually.. Can you measure if the motor drivers get 12V actually?

jsmithe2, 2024-07-17:
Yes, my terminal gets 12 volts just fine. One thing I did see was my "enable" pin on my A4988 driver gets a 3.3V read on my multimeter. I read online that if it draws that voltage, it will disable your driver and the steppers attached will not run? Are you familiar with this and do you know a solution to this?

jsmithe2, 2024-07-17:
@beniroquai Okay, I was able to solve the problem I just explained above by using a jumper cable to connect my enable and ground pins together which enabled all my drivers. The nema 11 motor I use for the z-axis works great in imswitch but the nema 17 motors I used for X & Y control just vibrate still with no real torque or movement. I am going to try to swap out the A4988 drivers with the more powerful DRV8825 drivers that can handle higher-rated currents. Hopefully that gets me the torque I need to rotate the motors instead of just shutting down because of overheating which I believe is happening right now. 

I'll get back to you on this but let me know if there is anything else I should potentially try? Is there a way to set the current limit in the firmware? do you think this would help?

beniroquai, 2024-07-18:
The enable pin is mapped to `PIN12` as taken from the youseetoo.github.io:

```

          // ESP32-WEMOS D1 R32
          const int PIN_DEF_MOTOR_DIR_A = 0;
          const int PIN_DEF_MOTOR_DIR_X = GPIO_NUM_16; 
          const int PIN_DEF_MOTOR_DIR_Y = GPIO_NUM_27;
          const int PIN_DEF_MOTOR_DIR_Z = GPIO_NUM_14;
          const int PIN_DEF_MOTOR_STP_A = 0;
          const int PIN_DEF_MOTOR_STP_X = GPIO_NUM_26;
          const int PIN_DEF_MOTOR_STP_Y = GPIO_NUM_25;
          const int PIN_DEF_MOTOR_STP_Z = GPIO_NUM_17;
          const int PIN_DEF_MOTOR_EN_A = GPIO_NUM_12;
          const int PIN_DEF_MOTOR_EN_X = GPIO_NUM_12;
          const int PIN_DEF_MOTOR_EN_Y = GPIO_NUM_12;
          const int PIN_DEF_MOTOR_EN_Z = GPIO_NUM_12;
          const bool PIN_DEF_MOTOR_EN_A_INVERTED = true;
          const bool PIN_DEF_MOTOR_EN_X_INVERTED = true;
          const bool PIN_DEF_MOTOR_EN_Y_INVERTED = true;
          const bool PIN_DEF_MOTOR_EN_Z_INVERTED = true;
```

So it should automatically go high/low whenever you activate the motor. Which string did you use for running the motor? 

Does this work?

```json
    {"task":"/motor_act",
        "motor":
        {
            "steppers": [
                { "stepperid": 1, "position": 10000, "speed": 5000, "isabs": 0, "isaccel":0},
                { "stepperid": 3, "position": 10000, "speed": 5000, "isabs": 0, "isaccel":0}
            ]
        }
    }
````

What are you trying to drive with the motors? 12V/3A + A4988 should give you plenty of torque. 
The firmware cannot set any current limit.

Farghal1, 2024-07-18:
> Can you check that pin without the CNC Shield connected? May also be that the ESP’s pin is fried.

Sorry for my late reply, but I had to try different ways to see where the problem is:
1- I tried to measure the output voltage on that pin with the CNC removed, but no output change.
2- I also tried a new ESP and CNC module and the problem is still the same.
3- The laser works fine separately and the board has 5v and 12v connected.

What else do you think could be the issue? what solutions do you suggest?

beniroquai, 2024-07-18:
And it shows UC2-WEMOS as pinconfig? I'm sorry, I don't really know what'S
going on then.. :(
Can you upload the simple blink example and change the gpio to e.g. 12 to
see if this works?

> Message ID: ***@***.***>
>

Farghal1, 2024-07-29:
Hii,

Sorry for this late reply. I uploaded the blink example and connected an Led to show the output. The Led blinks, but the laser doesn't. It's always ON and dissipating 0.28 Amp

Now Is this the Laser issue and should it be changed?
Or this is usual not to blink similar to the LED!

beniroquai, 2024-07-29:
Hmm, hard to say. What kind of laser do you have? Maybe a photo helps. You said that toggling the TTL signal from 5V to GND would turn off the laser. Perhaps it's indeed a driver issue. 
The STORM setup would also be without that. You anyway would need to bleach the fluorophores ;-) 
@AZehrer - right?

Farghal1, 2024-07-30:
This is the laser I'm using: 638nm Red Laser Module 500mW Round Dot Focusable TTL 3050
The same as the one here: https://www.laserlands.net/638nm-red-laser-module-500mw-round-dot-focusable-ttl-3050.html

beniroquai, 2024-07-30:
Could you provide a photo of your setup?

> Message ID: ***@***.***>
>

Farghal1, 2024-07-30:
[image]

This is the whole diagram. The laser has 3 pins (Blue, red, Black). red is connected to 12 v similar to the green in the pic. Blaack is connected to ground.  Blue is connected to spnEN. 
It's not complicated in terms of connections and what makes it more weired is the laser doesn't even blink when it's connected separately to the microcontroller board.

beniroquai, 2024-07-30:
did you swap 12V/Gnd on the input of the laser? As i said, there is a rectifier and hence a potential issue with a floating/virtual ground. 
Also, I meant a real image / photo of the system.

Farghal1, 2024-07-30:
Yes, When I swap the laser power input, the voltage goes to 5V and 3.7 Amps. However, the laser doesn't even turn on.

[image]

This is the actual connection for the laser. Blue -> SpnEn, Black -> GND, Red -> 12V

closed 2024-07-31 by beniroquai